pdbear/syno_nvidia_gpu_driver #1: DS3622xs+ Support?

cclarke99, 2023-10-05:
Does this package support the DS3622xs+? I'm following the instructions to use the physical GPU and when I try to run nvidia-smi, I get the following error.

/volume1/@appstore/NVIDIARuntimeLibrary/usr/bin# ./nvidia-smi
NVIDIA-SMI couldn't find libnvidia-ml.so library in your system. Please make sure that the NVIDIA Display Driver is properly installed and present in your system.
Please also try adding directory that contains libnvidia-ml.so to your system PATH.

Perhaps there are some path issues on this platform?

pdbear, 2023-10-05:
您好，您的邮件已收到，我尽快回复您！

cclarke99, 2023-10-06:
Seems that I missed a step and am making some progress now.

pdbear, 2023-10-07:
DS_AAAA_XS+ series not supported now.

closed 2023-10-17 by pdbear

ctaoist, 2023-12-06:
DS3622xs+ 7.1+ 版本的虚拟群晖求支持