pdbear/syno_nvidia_gpu_driver #7: VMWARE ESXI, Tesla P4 vGPU, DVA3221 i18n-arpl not allows AI/Face Recognition tasks.

hendymen, 2023-11-17:
Hello,

i am experiencing issues with the use of the latest driver (x86_64-510.108.03-common-gpuinfo-20230919.spk) on an DVA3221.

I'm able to see the vGPU inside the DVA3221 with DSM 7.2.1-69057 Update 2. But no AI tasks can be run.

[image]
[image]
[image]

sacredx72, 2024-04-16:
> Its either GPU model name is soft coded into Surveillance Station or Name of a package is wrong: Currently: ![IMG_20231128_143439](https://private-user-images.githubusercontent.com/60470490/286252311-7a9374cf-83d9-4da8-9916-e7183192a534.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyNjU0MTEsIm5iZiI6MTcxMzI2NTExMSwicGF0aCI6Ii82MDQ3MDQ5MC8yODYyNTIzMTEtN2E5Mzc0Y2YtODNkOS00ZGE4LTk5MTYtZTcxODMxOTJhNTM0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE2VDEwNTgzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4ZmI5YTAzOTkxNzcyNjFjMzliZDQ0N2RlZTg0NWE4MDYyZWQ1OTg2ZjMxNjM2MDYzYjNhOThiZmM5NWJiODcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.s4iFFcRLtUM02UUk0tKsQbJ5r8_2FpXPZqnbbAgZFLU)
> 
> Must be:
> 
> ![images (11)](https://private-user-images.githubusercontent.com/60470490/286252465-b5f5db30-8ac4-495f-bcda-e333e3b4362d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyNjU0MTEsIm5iZiI6MTcxMzI2NTExMSwicGF0aCI6Ii82MDQ3MDQ5MC8yODYyNTI0NjUtYjVmNWRiMzAtOGFjNC00OTVmLWJjZGEtZTMzM2UzYjQzNjJkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxNlQxMDU4MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NTBmYzkwNjhkN2EzOWZmODU4NjJiM2U3ZWI0M2ExZGRiNWRmMTViNzI5NmQ2YmE5Yzg0NTI0MjYwODY0YmIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.GhankV0f1x56Hqv3t9skDNCI89hqOHqCbIfFp2F7h2s)
> 
> Or use proxmox with gpu Passthrough to emulate gtx1650, gtx1660.
Good afternoon, can you tell me how to correctly simulate another card through Proxmox? I’m trying to change the PID/VID in the settings, but when I start Xpenolodgy, the video card is not detected, the PID/VID in the virtual machine remains 1234:1111

OrpheeGT, 2024-06-13:
I don't know if it can help, but as refered in another issue, it does not work for me either with a genuine 1650 wich works with standard Nvidia runtime library.

Looking in Surveillance Station libs, I searched for runtime references :

[image]

OrpheeGT, 2024-06-15:
So my tests : 

525 custom nvidia drivers is great for HW transcoding with Emby. the original 440.40 are too old for Emby sadly, and NVENC does not work with 440.44, whereas it works great with 525.

but it breaks DVA3221 Surveillance Station AI features. and you can't just uninstall modded and resintall original nvidia package.

You have to force a DSM reinstall (with keep all data and parameters). It works and you can then reinstall nvidia original package.

So it is up to the user currently to choose what he prefers, HW transcoding ? or Surveillance Station AI ? ...