pdbear/syno_nvidia_gpu_driver #10: DS1823xs+ support & T400 support

hyllm, 2024-01-01:
is this possible?

hyllm, 2024-01-05:
[image]

So... P400 is working with no problems... :)

hyllm, 2024-01-05:
[image]

hyllm, 2024-01-05:
No problems at all with Plex. :)

[image]

pdbear, 2024-02-09:
thanks alot

closed 2024-02-11 by hyllm