pdbear/syno_nvidia_gpu_driver #13: strange messages on 3622xs

kills, 2024-02-11:
Driver is installed and working fine, I can transcode in plex and see the streams in nvidia-smi. 👍 

Problem is now, my dmesg is going crazy with 1000's of entries:
[ 3640.393457] audit: type=1400 audit(1707656661.919:1709): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=17568 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0

Just wondering how to fix this?

kills, 2024-02-11:
[image]

kills, 2024-02-11:
[image]

kills, 2024-02-15:
I've been trying to look into this myself the past few days. So far I think we need an apparmor profile for nvidia-smi application on the system. But I'm not 100% sure. I am feeling a little out of my depth on troubleshooting this one, but will keep looking at it.

It appears to be related to accessing the folder /dev/char/195:0, /dev/char/195:255, /dev/char/245:0, /dev/char/245:1

as you can see in the logs here:
  240.517741] audit: type=1400 audit(1707971889.427:149): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=7365 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.742029] audit: type=1400 audit(1707971899.649:150): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:255" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.762008] audit: type=1400 audit(1707971899.669:151): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/245:1" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.781731] audit: type=1400 audit(1707971899.690:152): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/245:2" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.801457] audit: type=1400 audit(1707971899.709:153): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.821039] audit: type=1400 audit(1707971899.729:154): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[  250.842336] audit: type=1400 audit(1707971899.749:155): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=8036 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0

I am also seeing the following entries show up when I open Synology control panel Info page:
[ 1054.805977] audit: type=1400 audit(1707972703.664:624): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/195:255" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0
[ 1054.829290] audit: type=1400 audit(1707972703.687:625): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/245:1" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0
[ 1054.852237] audit: type=1400 audit(1707972703.710:626): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/245:2" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0
[ 1054.874906] audit: type=1400 audit(1707972703.733:627): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/195:0" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0
[ 1054.897623] audit: type=1400 audit(1707972703.756:628): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/195:0" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0
[ 1054.921835] audit: type=1400 audit(1707972703.780:629): apparmor="DENIED" operation="unlink" profile="/usr/syno/sbin/synoscgi//SYNO.Core.System.info" name="/dev/char/195:0" pid=10700 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=1 ouid=0

So there must be something still missing in the 3622xs release to avoid these messages from being logged

kills, 2024-02-15:
Or is it more likely the fact that the internal Synology snmp service is touching those folders and it doesn't have access to them?

I will explore this, but appreciate if anyone who might have more knowledge to make any suggestions :)

kills, 2024-02-17:
I've also found out that this only spams DMESG while I am logged into the Synology Web Console. If I remain logged out, no entries are found.

pdbear, 2024-02-24:
Ok，I will try to fix it next release.

pdbear, 2024-03-10:
fixed

closed 2024-03-10 by pdbear

JoveYu, 2024-04-18:
设备型号DS918+，矿神源中最新版本1.2.4-0309版本，驱动版本525.105.17，目前还是有这个问题，dmesg中大量audit信息
```
[756647.304412] audit: type=1400 audit(1713425014.486:43021): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:1" pid=18591 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756647.306650] audit: type=1400 audit(1713425014.488:43022): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:2" pid=18591 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756647.308735] audit: type=1400 audit(1713425014.491:43023): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18591 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756647.310769] audit: type=1400 audit(1713425014.492:43024): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18591 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756647.312812] audit: type=1400 audit(1713425014.495:43025): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18591 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.499043] audit: type=1400 audit(1713425024.681:43026): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:255" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.501280] audit: type=1400 audit(1713425024.683:43027): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:1" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.503402] audit: type=1400 audit(1713425024.685:43028): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:2" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.505544] audit: type=1400 audit(1713425024.687:43029): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.507664] audit: type=1400 audit(1713425024.689:43030): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756657.509848] audit: type=1400 audit(1713425024.691:43031): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=18703 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.494639] audit: type=1400 audit(1713425034.676:43032): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:255" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.496708] audit: type=1400 audit(1713425034.678:43033): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:1" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.498635] audit: type=1400 audit(1713425034.680:43034): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/247:2" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.500544] audit: type=1400 audit(1713425034.682:43035): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.502440] audit: type=1400 audit(1713425034.684:43036): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[756667.504499] audit: type=1400 audit(1713425034.686:43037): apparmor="DENIED" operation="unlink" profile="/usr/bin/snmpd" name="/dev/char/195:0" pid=19124 comm="nvidia-smi" requested_mask="d" denied_mask="d" fsuid=0 ouid=0
[757573.882231] audit: type=1101 audit(1713425941.064:43038): pid=7400 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:accounting grantors=pam_permit acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
[757573.884180] audit: type=1103 audit(1713425941.066:43039): pid=7400 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
[757573.886408] audit: type=1105 audit(1713425941.068:43040): pid=7400 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:session_open grantors=pam_env,pam_env,pam_permit,pam_unix,pam_limits acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
[757573.889070] audit: type=1110 audit(1713425941.071:43041): pid=7401 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
[757574.023254] audit: type=1104 audit(1713425941.205:43042): pid=7400 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
[757574.025245] audit: type=1106 audit(1713425941.207:43043): pid=7400 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:session_close grantors=pam_env,pam_env,pam_permit,pam_unix,pam_limits acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'

```