pdbear/syno_nvidia_gpu_driver #14: Unable to view long term GPU usage history

cclarke99, 2024-03-02:
First of all, thank you for adding support for XS+ systems!

An issue I'm running in to on my DS3622XS+ is that I cannot view historical GPU usage info. I can only see the graphs for "realtime." Selecting any of the historical times from the drop-down menu results in the following error.

[image]

KennyH, 2024-05-04:
+1

My setup reproduces this as well. I suspect it isn't implemented yet.

pdbear, 2024-07-19:
I will check it in next version.

pdbear, 2025-06-22:
I forgot to check this part in this version. I just finished working on the 30-series unlock recently, so I'll take a look when I have time later. 😂