pdbear/syno_nvidia_gpu_driver #19: Please Help me check my setup

keitetran, 2024-04-21:
My spec same above image. 
I have success install this driver. But on resource monitor GPU not working.
i can use this driver ? What i am missing ? 

Thank you verry much.

[image]
[image]

KennyH, 2024-05-04:
I am not associated with this project, but saw your issue when I was posting my feature request. 

Two guides that may help you troubleshoot or try different things:

1. [Guide: How to add a GPU to Synology DS1820+](https://www.reddit.com/r/synology/comments/16vl38e/guide_how_to_add_a_gpu_to_synology_ds1820/) -- covers setup and docker pass-through for Plex
2. [XPEnology Community - Nvidia Runtime Library](https://xpenology.com/forum/topic/22272-nvidia-runtime-library/) -- a long discussion on this topic that predates this project, but references it on the 5th page.

Also, if you would like help with your setup, more information about your issue would need to be provided. What is your system and GPU? How is Plex and Emby running on your system? Docker only or through spk? Stuff like that.

Wish you luck!

keitetran, 2024-05-05:
Thank for comment. 
I tried it and success get GPU working now. 

My hardware: 
Motherboard: Quanmax MITX-DNV6
CPU Intel(R) Atom(TM) CPU C3538 @ 2.10GHz
RAM 16Gb
GPU NVIDIA GeForce GTX 1050 Ti
Bootloader: Arc (or RRorg)
DSM 7.2.1-69057 Update 5

Sortware:
Plex media server install from store
Plex Pass: Actived

Changelog: 
1. Install latest driver version https://github.com/pdbear/syno_nvidia_gpu_driver/releases
2. Execute command
```
# Command not found
nvidia-smi

# Fix 
vgpuDaemon fix
vgpuDaemon stop
vgpuDaemon start

# Try again (It worked)
nvidia-smi
```

3. Install patch from https://github.com/keylase/nvidia-patch

DONE !!! THANK YOU VERY MUCH 

[image]
[image]

closed 2024-07-19 by pdbear