pdbear/syno_nvidia_gpu_driver #21: Feature Request: Update for Cuda 12.1 or higher version

KennyH, 2024-05-04:
First, I want to say that I appreciate this work immensely! You have done a wonderful job and I was very excited to try it out on my Synology DS1823xs+ with a T400 when I first learned about your work.

My request is to see if it were possible to update to have at least Cuda version 12.1 enabled. Currently, it is 12.0. 

I am requesting this because I would like to have a gpu-enabled instance of handbreak ([handbrake-nvenc-docker](https://github.com/zocker-160/handbrake-nvenc-docker)). And, unfortunately when I run it, it fails and says that it needs version 12.1 or higher and only 12.0 is detected.

Thank you for your consideration.

pdbear, 2024-07-19:
ok, latest version update to cuda 12.2

closed 2024-07-19 by pdbear

KennyH, 2024-07-20:
THANK YOU!

This is perfect now!