pdbear/syno_nvidia_gpu_driver #28: Package Install issue with RackStation RS18016xs+

maxwdjoh, 2024-12-10:
I was trying to install the package onto my RackStation with the hope of getting a good transcoding card working. The package center will not allow me to install the package. Is there any major reason on why it would not work? It is running DSM 7.2.1. I have added some photos of the info for this model. 

[image]

[image]

pdbear, 2025-07-16:
Can you log into SSH, enter the following two commands, and provide the output?

uname -r
cat /proc/syno_platform

paulbhome, 2026-01-04:
> Hi, Did you get anywhere with this?
>
> Unfortunately no. I was able to get the Synology software to show a GPU type but never got it to display a GPU. The package would kind of install but the driver would never load.

Thanks